n0bel/PiClock #230: PiClock The Next Generation:  PiClock3

n0bel, 2021-12-12:
Up until now PiClock has been a very monolithic program.

I've started a completely rewritten PiClock at the repository https://github.com/n0bel/PiClock3

This version is driven by yaml configuration files and modules.

The configuration file controls the display and the pages.   Modules are then used to place content on the page.

There's no documentation, you'll need to infer how it works from the source and the config files.  Currently only clocks and METAR current conditions are implemented.

Give it a try and report back what you think.

https://github.com/n0bel/PiClock3  some instructions for those willing to dare the command line are there.

lstew452a, 2022-01-09:
Will PiClock 3 replace the current PiClock. I tried PiClock3. Works fine.

n0bel, 2022-01-10:
For a couple years, after PiClock3 is implemented, I'll probably keep the original PiClock going.

n0bel, 2022-01-17:
Please note that PiClock3 is only partially complete.  
I've updated the page to reflect current status. https://github.com/n0bel/PiClock3   
Things are progressing.  Any contributions are welcome.

SerBrynden, 2026-08-24:
@n0bel Will PiClock3 also have the ability to allow users to add an optional map layer overlayed on top of the radar layer, similar to https://github.com/SerBrynden/PiClock ?  I find it very helpful to have a semi-transparent layer with just the state boundaries, city names, roads, etc. on top of the radar layer so, when the radar window is completely filled with greens, yellows, and reds from the large storm, I can still see the map locations for geographic reference.  Otherwise, I just have a radar window filled with colors.  I use customized map styles from MapBox for the base map and semi-transparent overlay map.

n0bel, 2026-08-24:
i'll see if the maploop plugin can add a transparancy  alpha factor to the overly

n0bel, 2026-08-24:
And the forecast column is no longer empty.

`classic` has always reserved nine cells down the right hand side for a
forecast, and until now they were nine empty boxes.  There is an
**Open-Meteo** provider behind them, which needs no key.

More usefully, reading the weather is now separate from drawing it - the
same split the radar has had.  A weather provider answers three questions:

```
current()          what it is doing now
hourly(n, step)    the next few hours
daily(n)           the next few days
```

and answers honestly when it cannot.  A METAR station reports what is
happening rather than what will, so it returns empty for the last two, and
pointing a forecast at one draws nothing rather than a guess.

Two widgets draw whatever they are given:

```yaml
providers:
  metar:     {plugin: PiClock3.Metar, METAR: KLVN}
  openmeteo: {plugin: PiClock3.OpenMeteo}

widgets:
  current-conditions:
    plugin: PiClock3.CurrentConditions
    conditions-provider: metar        # or openmeteo
  forecast:
    plugin: PiClock3.Forecast
    forecast-provider: openmeteo
```

So the conditions block can be a real observation from the airfield down the
road, or a model where there is no airfield within useful distance, and
swapping is one line.  I have both running side by side here; they disagree
about the sky often enough to be interesting - the station says CLR at the
field while the model says partly cloudy for the grid square, and both are
right.

The forecast column follows v1's shape: three hourly cells, then six daily,
picture on the left, conditions in words over a line of figures, the day in
the bottom corner.

**This breaks configs again** - the second time in two days, and I would
rather it were the last.  Anything naming `PiClock3.Metar` as a widget now
wants it as a provider with a `CurrentConditions` widget in front.
`BREAKING-CONFIGURATION-CHANGE-2026-08-23.md` shows both forms side by side.
Starting again from `Config-Example.yaml` gets it all wired.

Open-Meteo asks for no key but its data is CC-BY, so the provider declares
an attribution and the widget prints it at the foot of the column.  A
station's attribution is its id, which is what the conditions block already
showed beside the observation time.

## Where that leaves it

**PiClock3 is now back to the base version of PiClock.**  Everything the
original does on screen, this does: clock, date, almanac, current
conditions, radar, forecast.

What it does not have yet is the breadth:

- **Providers.**  PiClock has four weather sources - METAR, Open-Meteo,
  OpenWeatherMap and Tomorrow.io.  PiClock3 has the first two.
- **Languages.**  PiClock ships German beside English; PiClock3 has one
  `Language-en.yaml` and everything else falls back to it.

That breadth is the next thing to tackle, and it should be the easy part -
a provider is a plugin answering three questions, and a language is a file
of words.  Neither needs the architecture to move again, which is rather
the point of having spent two days moving it.

n0bel, 2026-08-24:
Paths in my last comment have changed - the examples moved to `examples/`
and dropped the word:

```
cp examples/default.yaml Config.yaml
cp examples/ApiKeys.yaml ApiKeys.yaml

python3 PyQtPiClock3.py examples/meadow.yaml
```

The folder is the point: a plugin can ship `themes/`, `layouts/`, `units/`
and `examples/` beside its code and arrive complete.  Core follows the same
convention rather than asking others to.

n0bel, 2026-08-24:
`WeatherCommon` is gone.

It had become the thing everything reached into - a forecast widget had to
ask a *weather plugin* to format a temperature.  Units are core now: a table
of data rather than an if/elif chain.

```yaml
units: metric        # or default, SI, nautical
```

The table lives in `PiClock3/units/`, found the way themes and layouts are
found - a `units/` folder of your own, or a plugin's, merges over it.  Each
quantity names a base unit and every other is a factor and an offset, so
adding one converts to and from everything else for free.  `prefix` and
`suffix` belong to the unit, because a currency wants its symbol in front
and a temperature wants it behind.

Quantities are named for what wants a different unit, not for dimension:
`depth`, `distance`, `altitude` and `height` are all length and none want
the same one.  `pressure` and `altimeter` are separate for the same reason -
one is what a barometer reads, the other is corrected to sea level, and
mixing them silently is how a reading comes out 38mb wrong.

Two things fell out of it.  Providers now hand over numbers and widgets only
format, so a station computes its own humidity and how it feels rather than
leaving it to whatever draws it.  And day or night comes from actual sunrise
and sunset instead of `hour >= 18 or hour < 6` - which was wrong by hours in
summer, and wrong for half the year anywhere near the poles.

**Configs change again**: delete the `weather-common:` provider, and add
`units:` if you want anything but US units.  That is the third break in
three days and I am not pleased about it either - but this one deletes a
line rather than moving one.

n0bel, 2026-08-25:
**Languages, and conditions said in notation**

A language is now one file — the codes it answers to, the system locales that mean it, the words, and a table of weather conditions. It's found on the same search path as themes, layouts and units, so a `languages/` folder of your own or a plugin's merges over the shipped one. `en` and `de` ship; `examples/berlin.yaml` is the German clock.

Providers stopped speaking English. They say what the sky is doing in **WMO code table 4678 notation** — the codes a METAR is written in — and the widget turns that into words in whatever language the clock is set to. `Metar.py` contains no English at all now. Cloud amount has no 4678 notation, so the METAR sky-cover codes carry it.

Four bugs turned up on the way:

- METAR lists present weather in *decreasing* significance, and I was reading the last group. `-RA BR` displayed as "Mist", `+SN BLSN` as "Blowing Snow".
- `CAVOK` matched no sky group, so a clear evening in Berlin drew a blank description and a cloudy icon.
- Snow accumulation showed Open-Meteo's liquid-water equivalent labelled "Snow" — roughly a seventh of the actual depth. `snowfall` is asked for separately now, in cm.
- `LC_TIME` was set from the Date widget's tick, racing the forecast's network callback for who got to name the days. Set once at startup now, from the locales the language file lists.

Providers also carry `raw` — the service's own record for an entry, un-normalised — so nothing is lost by normalising the rest.

The provider contract is written up in CONTRIBUTING.md for anyone wanting to add a source.

n0bel, 2026-08-25:
**Docs caught up with the code**

The provider contract is written down now, in CONTRIBUTING.md. Anyone adding a weather source had no way to learn it must say what the sky is doing in WMO 4678 notation rather than in English — which is the difference between a source that can be translated and one that can't.

Auditing README and CONTRIBUTING against the code turned up one claim that was simply false: wind and humidity in the forecast column, described as already carried by the provider. They are in current conditions only — the hourly and daily grids never asked Open-Meteo for them. Removed rather than reworded. Everything else checkable was checked: the eleven icon names against the files on disk, the unit sets, the api key names, the example coordinates and station, every plugin named in the tables, and the dependency versions against PyPI.

Also closed [#5 on PiClock3](https://github.com/n0bel/PiClock3/issues/5), reported in 2022 — a config with no `styles:` block died at startup. It is guarded now, and better than guarded: styling moved into themes, so no shipped example has a `styles:` block at all any more, and every run takes the path that used to crash.

n0bel, 2026-08-25:
**This thread continues over in PiClock3**

Progress, plans and half-formed ideas now live at https://github.com/n0bel/PiClock3/issues/11, in PiClock3's own repository, where they belong now that it has one. That thread carries what works today, what is on the list, and the sources being looked at but not promised.

This issue stays open — PiClock v1 is still going, and this is still the place for anything about it.

For either project, a bug or a specific request is better as its own issue than a reply on a long thread: it gets answered sooner, stays findable, and can be closed when it is done.

lstew452a, 2026-08-25:
I do not use google Api. Is there any other option like the old PiClock. Larry

Progress, plans and half-formed ideas now live at n0bel/PiClock3#11<https://github.com/n0bel/PiClock3/issues/11>, in PiClock3's own repository, where they belong now that it has one. That thread carries what works today, what is on the list, and the sources being looked at but not promised.