n0bel/PiClock #231: Getting error periodically

ScottChapman, 2021-12-16:
Every once in a while my clock just stops working. What I've seen in the logs looks like:
```
https://api.darksky.net/forecast/776b55d1b8978d8d13e251fcc8e78e59/42.5911248,-71.5692887?units=us&lang=en&r=0.232346411801
('wxmetar', '')
Traceback (most recent call last):
  File "PyQtPiClock.py", line 1117, in wxfinished_metar
    dt = f.time.replace(tzinfo=tzutc()).astimezone(tzlocal.get_localzone())
AttributeError: 'NoneType' object has no attribute 'replace'
Traceback (most recent call last):
  File "PyQtPiClock.py", line 587, in wxfinished_ds
    wxdata = json.loads(wxstr)
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
```
Any ideas?

n0bel, 2021-12-16:
Since both Metar and darksky are being affected by similar errors, it means something is interfering with http responses on your internet.   Empty responses are being received.  These could be timeouts, or router issues, or wifi issues.

lstew452a, 2022-01-12:
After I installed PIClock on my Raspberry PI running Raspberry Desktop when I  tried to run Chromium version 95.0.4638.78 running on Raspbian 11 (32 bit). I tried all the steps to resolve the probelm to no avail.
Using Firefox it also crashes I tried ti Restore This Tab to no avail.

I am running a Raspberry PI 400 with a  Seagate External harddrive with 1 TB to the USB 3.0 connector.

Every thing fine unitl I installed Scott's version of PIClock. I thought this was interesting and be related.

n0bel, 2022-01-12:
PiClock has nothing to do with browser issues.  Its not related.

closed 2022-02-13 by n0bel