n0bel/PiClock #234: First run failing after fresh install

vanhalen26505, 2022-01-17:
Hi. I have installed and run PiClock numerous times in the past. I decided to do a fresh install today and the initial run using **sh startup.sh -n -s** fails under both Stretch and the most recent release of Raspbian. I do not know enough to troubleshoot this error further and would love some guidance:

Traceback (most recent call last):
  File "PyQtPiClock.py", line 8, in <module>
    import dateutil.parser
ImportError: No module named dateutil.parser

Thanks very much in advance,
Lee

SerBrynden, 2022-02-13:
You need to install more needed python libraries:
`pip install python-dateutil --upgrade
pip install tzlocal --upgrade
pip install python-metar --upgrade`

vanhalen26505, 2022-02-13:
Will give it a shot, thanks. Was this included in the directions and I
potentially missed it?

GrampDamnpus, 2022-03-06:
I can say I followed the directions and did the upgrades but still couldnt get it to work, I finally had to use your fork and it worked. 

> You need to install more needed python libraries, per the PiClock install instructions:
> 
> ```
> pip install python-dateutil --upgrade
> pip install tzlocal --upgrade
> pip install python-metar --upgrade
> ```

closed 2022-03-06 by n0bel

ScottChapman, 2022-04-07:
I am running into this on bullseye. Any ideas?