n0bel/PiClock #236: Temp not updating?

ScottChapman, 2022-02-12:
So, I noticed recently, like a couple of days ago, that the temperature does not appear to be updating.

But... It kinda looks like things are updating OK, like the forecasts on the right appear to be advancing.

I took a quick scan through the logs and didn't see any errors (except for an occassional index out of range when I think it is doing something with images. But that seems different...

When I look at my darksky dashboard I don't see any calls in last couple of days...

n0bel, 2022-02-12:
I have one PiClock still using DarkSky.  Mine is updating normally.   My guess would be your ISP or somewhere there's a caching proxy in the middle thats defeating the random number querystring.

ScottChapman, 2022-02-12:
anything I can do to test that? Maybe add some logging to where that is call is taking place and dumping some stats?

ScottChapman, 2022-02-12:
OK, it looks like there was just some lag in darksky updating their dashboard metrics, I do see pretty consistent calls every day. And I did test the URL I saw in the logs and got correct current forecast. So I think that's not it.

To be clear, it is only the box in the upper left that appears to be stale. Everything else looks like it is updating OK.

Can you point me to the code which is responsible for that and perhaps I can instrument it to get some idea.

ScottChapman, 2022-02-12:
Picked the next nearest station. All set!

closed 2022-02-12 by ScottChapman