n0bel/PiClock #241: Radar has lines across the maps.

lstew452a, 2022-06-06:
Has anybody noticed this?

feh123, 2022-06-06:
I do, very occasionally, see diagonal broken lines but the rest of the radar image seems correct. They usually clear themselves after around 1 - 2 hours so I think it's a corrupted download. I am using the Serbrynden fork.

taftr1, 2022-08-25:
I see these constantly....   What it seems like is the downloaded images that are making up the moving image are not aligning properly...

closed 2023-04-02 by n0bel

SerBrynden, 2023-04-15:
@n0bel Finally solved!  When constructing a QImage object, Qt offers this warning:

"Warning: This will create a QImage with uninitialized data. Call fill)() to fill the image with an appropriate pixel value before drawing onto it with QPainter." https://doc.qt.io/qt-5/qimage.html#QImage-2

Under `def combineTiles(self)` after `ii = QImage(self.tilesWidth * 256, self.tilesHeight * 256, QImage.Format_ARGB32)` insert `ii.fill(Qt.transparent)` to solve everything.