n0bel/PiClock #247: Adding NWS Alerts to the clock

xxxDrewedxxx, 2023-03-29:
Using various methods of stealing code from other peoples projects I have bashed together the following code.
`import requests
import json
import time

# Define your latitude and longitude
latitude = YOUR_LATITUDE
longitude = YOUR_LONGITUDE

# Define the NWS API endpoint and request parameters
endpoint = f"https://api.weather.gov/alerts/active"
params = {
    "point": f"{latitude},{longitude}"
}

# Define a function to print the active alerts
def print_alerts(alerts):
    for alert in alerts:
        print(f"{alert['properties']['event']} - {alert['properties']['headline']}")

# Loop continuously to check for updates every 5 minutes
while True:
    # Send a GET request to the NWS API endpoint
    response = requests.get(endpoint, params=params)
    
    # Check if the request was successful
    if response.status_code == 200:
        # Parse the JSON response and extract the alerts
        data = json.loads(response.text)
        alerts = data['features']
        
        # Check if there are any active alerts
        if len(alerts) > 0:
            # Print the active alerts
            print_alerts(alerts)
        else:
            # If there are no active alerts, print a message
            print("No active alerts.")
    else:
        # If the request was unsuccessful, print an error message
        print(f"Error: {response.status_code}")
    
    # Wait for 5 minutes before checking for updates again
    time.sleep(300)
`
Will this work? and if so can I just drop it in PyQtPiClock.py?  I'd have to figure out how to get it to display where I want, and I'm not sure if it would break any of the other timers (weather and radar?)

xxxDrewedxxx, 2023-03-29:
And evidently I can add code for crap...

xxxDrewedxxx, 2023-03-30:
That's not a problem for me, I don't actually use the clock feature.  I working on changing the layout to be weather first.  eventually I'd like to pull in data from my own weather station ( via API ) but that is down the road.
This is work in progress, but you can get the idea.
[image]

xxxDrewedxxx, 2023-03-31:
> The OWM API also includes alerts for your location, and in a much easier way. ...

Ah!  I see that is included on "One Call"  I'll have to sign up for that and dig into that...

xxxDrewedxxx, 2023-03-31:
@SerBrynden   Sorry to bug, but I'm beating my head against this wall.  Can you help out?
andrewfeldner@gmail.com

SerBrynden, 2023-04-01:
> @SerBrynden Sorry to bug, but I'm beating my head against this wall. Can you help out? [andrewfeldner@gmail.com](mailto:andrewfeldner@gmail.com)

@xxxDrewedxxx I can try. What's up? How far have you gotten?

xxxDrewedxxx, 2023-04-02:
@SerBrynden Honestly.  I haven't gotten anywhere.  It should be easy as API 3 already has them, but I really don't know what I'm doing.

xxxDrewedxxx, 2023-04-09:
Done.  This can be closed

closed 2023-04-09 by xxxDrewedxxx

vanosg, 2023-06-18:
Are you able to share your code for this? This feature looks very interesting

SerBrynden, 2023-06-22:
The only reason I haven't implemented weather alerts (other than not knowing where to place them on the screen) is you would have to fetch the alerts very frequently in order for them to be timely and relevant.  If the script is fetching alerts every 10 or 30 minutes, you could miss a tornado warning, for example, in which seconds matter.  I’d rather just get warnings from my weather radio or mobile phone.  The NWS might get annoyed if you’re calling their API every second, and the OpenWeather API (which also includes alerts) has obvious call limits.

xxxDrewedxxx, 2023-06-22:
> Are you able to share your code for this? This feature looks very interesting

I have this working on my PiClock.  It should work if you use my master:
https://github.com/xxxDrewedxxx/PiClock

You might have to tweak the layout to fit your screen
I think I have my set to ping for alerts every 5 min.  Which is slow on tornado, but more than quick enough for freeze warnings and red flag warnings.
[image]