n0bel/PiClock #249: tempf2tempc function missing

BertLindeman, 2023-04-02:
Running the current PyQtPiClock from the master repo with the Config-Example-Berlin 
Probably related to issue #185

```
Traceback (most recent call last):
  File "PyQtPiClock.py", line 1050, in wxfinished_tm2
    s += '%.0f' % tempf2tempc(f['values']['temperature']) + u'°C'
NameError: global name 'tempf2tempc' is not defined
Traceback (most recent call last):
  File "PyQtPiClock.py", line 1150, in wxfinished_tm3
    '%.0f' % tempf2tempc(f['values']['temperatureMin']) + u'°C'
NameError: global name 'tempf2tempc' is not defined
```

Also seen:
```
  File "PyQtPiClock_newissue.py", line 1152, in wxfinished_tm3
    s += Config.LRain + '%.0f' % inches2mm(paccum) + 'mm/hr '
NameError: global name 'inches2mm' is not defined
```

Added similar functions near line 300, and tested OK, but probbly there are now duplicates.

```
def tempm(f):
    return (f - 32) / 1.8

def tempf2tempc(f):
    return (f - 32) / 1.8

def inches2mm(f):
    return f * 0.621371192

def speedm(f):
    return f * 0.621371192

```

Maybe related: The forcast squares on the right side have English text not translated to German.

SerBrynden, 2023-04-02:
n0bel copied my functions for Tomorrow.io but didn't test it to notice that I renamed the functions for unit conversions. ;-)  

Not sure about the language translation yet. I'm on the road right now and can't take a look.

n0bel, 2023-04-02:
fixed all these issues in https://github.com/n0bel/PiClock/commit/6080cd9242882bc4576593c2db5a5eacdaae0251

BertLindeman, 2023-04-02:
Speedy Gonzales 😉

closed 2023-04-02 by BertLindeman