n0bel/PiClock #251: Is your weather information gone?  DarkSky Was Shutdown March 31, 2023.

n0bel, 2023-04-03:
There has been lots of discussion of this over the last 2 years.  See Issue #185 

Because of the shutdown, a new weather API is needed.   Current choices supported by PiClock are OpenWeatherMaps or Tomorrow.io.  You'll need to sign up to one of those services.

The fix is to review and perform the update process as shown in https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#updating-to-newerupdated-versions   Depending on other changes you may have made, you may want to start over, in which case starting at the top of that document may be in order.

update.py will ask for an OpenWeatherMaps api key.   If you wish to use Tomorrow.io, you'll need to manually your ApiKeys.py.  Remember that only one of owmapi or tmapi may be used.  Leave the opposite one commented out.   Read the comments in ApiKeys-Example.py

I'm closing #185   If you wish to report new problems, create a new issue.  However, reviewing #185 may be helpful to you.

SerBrynden, 2023-04-11:
> It looks as if pyqt4 is not available to install anymore. Has anyone had any luck with it on a fresh build?

@skipwill You have to use an older version of Raspberry Pi OS or Raspbian that supports Python2 in order to use PyQt4. Otherwise, try my fork which is written with Python3 and PyQt5. https://github.com/SerBrynden/PiClock

skipwill, 2023-04-22:
Hi Brendan,

Just a quick note to thank you for your fork of PiClock to pyqt5 and Python 3. It’s working great and I am thrilled to have it running again. Additionally I was successful in adding my tide clock hand code, to the analog click.

Again, thanks and all the best!

Skip Williams

vanosg, 2023-06-20:
@skipwill Hi again, I was still hoping you could share your tide clock hand code? I would love to implement something like that into my own instance. Would you mind posting it here or to something like pastbin.com ? Thanks!

skipwill, 2023-06-20:
Hi Geo,
I am using Brendan’s fork of PiClock so not sure if I should post it. I’ll be glad to send you a copy with the tide click code (it ain’t real pretty :-) ) Along with the ‘tide hand’ in a zip file if you would like.

Skip

SerBrynden, 2023-06-20:
Skip could create their own fork of PiClock and upload the changes there.

skipwill, 2023-07-09:
Geo,

Sorry for the delay. Things have been real busy here.

Attached is the source of PyQtClock.py with the tide hand.  All you should need to do is add

tidehand = 'images/tidehand.png'
tidestation = '8661070'  #NOAA tide station

to your config.py file.

Let me know if you have any problems with it. Some mail clients and/or antivirus programs  have a problem with .zip files.

Skip

vanosg, 2023-07-09:
Thanks, but nothing attached- could you edit via the Web UI and re-attach, or maybe create your own fork with the files? Thanks so much for your dedication

skipwill, 2023-07-10:
It might be on GitHub now. See if you can find it  if not let me know. Might be under user Skipwill. Let me know if it works!

Skip

SerBrynden, 2024-04-18:
OpenWeather One Call API 2.5 to be closedin June 2024: 
https://openweathermap.org/one-call-transfer

Knackster, 2024-04-18:
We are having to deal with this at work right now. Going from 2.5 to 3. I
was working with my web developer to try to figure out exactly what we were
going to do. If we're going to either upgrade or move to another service.
If we moved to another service, I'll update the group to what we go with.
The software that we write for law enforcement incorporates 2.5 to give
localized weather to patrol officers while on on patrolling their cars.