n0bel/PiClock #255: Not a issue per say

maserowik, 2023-04-10:
Upgraded to the latest version of PiClock.  Is there a way to get the month and date to be to the left fo the day?
[image]

Thanks 
Mike

xenon462, 2023-04-10:
in the PyQtPiClock.py file at the very end of AlignRight fix AlignLeft
[image]

maserowik, 2023-04-11:
Looking to add the date also
for example
4/10/2023 Monday
4/11/2023 Tuesday

xenon462, 2023-04-12:
file PyQtPiClock.py line 527 (three-hour weather forecast. First, second and third line in a column)

day.setText("{0:%A %I:%M%p}".format(datetime.datetime.fromtimestamp(
change

day.setText("{%m %d %Y 0:%A}".format(datetime.datetime.fromtimestamp(
or 0:%A %H:%M%p

file PyQtPiClock.py line 590 (daily weather forecast. Fourth - ninth line of the whole column)

day.setText("{0:%A}".format(dt))
change
day.setText("{%m %d %Y 0:%A}".format(dt))
or 0:%A

more about the time format
https://docs.python.org/3/library/datetime.html