n0bel/PiClock #256: Date strings not always in locale

BertLindeman, 2023-04-10:
The localized date and times are not always translated, 
see [in issue 185](https://github.com/n0bel/PiClock/issues/185#issuecomment-1493421518)

I think I found a reason. 
In the forcast squares before the day.setText calls I added a setlocale and some prints.

Before and after the setlocale it shows:
```
BEFORE: in wxfinished_tm3 ('en_US', 'UTF-8') default: ('en_US', 'UTF-8')
AFTER   in wxfinished_tm3: ('nl_NL', 'UTF-8') default: ('en_US', 'UTF-8')
```

So each time e.g. wxfinished_tm3 is called the locale is again back to the default.

The setlocale is done in the function Tick.
I assume the function `tick()` runs as a separate thread (or something like it)
Could that be the reason that *sometimes* I saw the correct locale datea and most of the time *not*.

I copied the setting of the locale near line 2200 between:
```
width = rec.width()

signal.signal(signal.SIGINT, myquit)
```

I added there:
```
# ================== Set locale here to test if the dates/times ARE in the locales:

if Config.DateLocale != "":
    try:
        locale.setlocale(locale.LC_TIME, Config.DateLocale)
    except Exception as e:
        print("In Tick setlocale failed:", e, "\n\t", Config.DateLocale)
        raise SystemExit(1)

```

Could this be the cause? 

In python3 I did not see a similar problem, but that can be a timing issue aswell.

SerBrynden, 2023-04-10:
What happens when you move the following lines from the `tick()` function to the `qtstart()` function?  Does that fix the issue?
```
if Config.DateLocale != "":
    try:
        locale.setlocale(locale.LC_TIME, Config.DateLocale)
    except:
        pass
```

BertLindeman, 2023-04-10:
Setting locale more often might easily miss some place in the code where it should be. (not formulated very legible, sorry)
But doing it on the one location (and a correct one) would be better.

Another side remark: I did not see this problem running python3. 
But not with this excact source due to print incompatibilites.

SerBrynden, 2023-04-10:
I saw the same problem in my Python3 fork, until I moved the code to qtstart().

n0bel, 2023-04-10:
pull request?  or should I just make the change?

BertLindeman, 2023-04-10:
I want to try.

BertLindeman, 2023-04-10:
Not doing enough pull requests lately.

Created the pull request. Not sure if I now should close the pull request (now)

SerBrynden, 2023-04-19:
My latest pull request #258 also fixes this.