n0bel/PiClock #260: Not so much an issue as a call for help with crontab

feh123, 2023-06-16:
I have been trying to reboot my PiClock on a schedule with crontab. My code is:

01 19 * * * /sbin/reboot
@reboot sleep 240 && /home/pi/PiClock/startup.sh

If I just reboot the Pi I only get to the Bullseye desktop on my monitor. So this code was my attempt to get the Pi software to re-install and then run startup.sh.

I have tried other variations like adding sh /home/pi/PiClock/startup.sh. I am missing something because the cronjob never works. Any ideas greatly appreciated.

SerBrynden, 2023-06-16:
Did you try either of these methods: 
https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#setting-the-clock-to-auto-start

SerBrynden, 2023-06-16:
Also, per the instructions, the auto-reboot setup should be done as root
`sudo crontab -e`
add the following line
`01 19 * * * /sbin/reboot`

But the PiClock auto-start should NOT root
`crontab -e`
add the following line
`@reboot sh /home/pi/PiClock/startup.sh`

feh123, 2023-06-16:
And it worked brilliantly. Thanks!

feh123, 2024-01-09:
Thanks @SerBrynden

closed 2024-01-09 by feh123