n0bel/PiClock #261: A few issues...

duskglow, 2023-06-17:
Hi, I just managed to get this running on my system, but I encountered a few issues.  Am happy to PR code back, but you may not be able to use it as is.  So, feedback needed.

1) On the latest version of rasbpian, it uses qt5 instead of qt4.  So the code is broken.  I was able to port it for the most part, but I may have encountered some issues of which I'm not aware yet.  Would you like a PR for that or would you like to discuss how to properly integrate that, as it probably still works for previous versions?
2) On the latest version of raspbian, a version of python is used that breaks your code.  Specifically, you can't use print without parentheses anymore.  I fixed that in my code.  That I can PR up if you want it easily.
3) Python is not a great language for catching errors, and there's not a lot of good error handling code.  I guess that makes perfect sense for a project written by someone who just wants to get something done, but makes for lots of crashes if there are issues.  Maybe I'll work on that a bit in my Copious Free Time (tm)
There's some functionality that I want, and I'll probably need to pull directly from the NWS feeds to get it.  Want a PR for that when I'm done?

Thanks, this at least lets me kickstart what I wanted to do.  Now to figure out how to turn the screen saver off...

SerBrynden, 2023-06-17:
Try this fork https://github.com/SerBrynden/PiClock
It works with the latest Raspberry Pi OS, Python3, and PyQt5.  There are pulls requests in the queue here, but the original author hasn't approved them yet.

duskglow, 2023-06-17:
Okay.  My tweaks are working for now and there are other things I want to do as well (like an overlay when there's a severe weather warning), so I'll use my fork for now.

SerBrynden, 2023-06-22:
The only reason I haven't implemented weather alerts (other than not knowing where to place them on the screen) is you would have to fetch the alerts very frequently in order for them to be timely and relevant.  If the script is fetching alerts every 10 or 30 minutes, you could miss a tornado warning, for example, in which seconds matter.  I’d rather just get warnings from my weather radio or mobile phone.  The NWS might get annoyed if you’re calling their API every second, and the OpenWeather API (which also includes alerts) has obvious call limits.

SerBrynden, 2023-11-15:
Recommend closing this issue.

henkdegroot, 2023-12-29:
> Try this fork https://github.com/SerBrynden/PiClock

Thanks for this, just installed PiClock on my Pi and hit the same issues. Using your fork solved most of it.

One more thing I noticed is that the openweather API now is 3.0 (instead of 2.5). As I had to create my API key recently, the one I created cannot use the 2.5 One Call api (it fails with a message the api key is not valid). The same onecall to 3.0 works fine.
At first I did not noticed this issue existed until I checked the logs, as when One Call fails, it is using some other api calls.
I have updated my local python script to use the 3.0 url and it works without any error message now.
Maybe an issue needs to be raised for this?

SerBrynden, 2023-12-29:
PiClock was updated earlier this year when OpenWeather released the 3.0 One Call and changed their API subscription model.  New PiClock users found out that when they signed up for a new OpenWeather API key, it didn't work because PiClock was only using the 2.5 One Call API calls.  But older OpenWeather accounts could still use the 2.5 One Call.  It was also discovered that if new users signed up for the 3.0 One Call (which is also free but requires a credit card in case you have overages), that new API key works with both the 2.5 and 3.0 One Call.

So PiClock was updated to first try the 2.5 One Call and, if that fails, it falls back to using the two basic free OpenWeather calls, one for current weather conditions and one for forecast.

If you signed up for the 3.0 One Call, it should work with the 2.5 One Call that PiClock uses.  You just have to wait a couple hours for it to be active in OpenWeather.  I just signed up today for a new OpenWeather account for testing, and a 3.0 One Call API key does still work with the 2.5 calls.  There seems to be no need to update PiClock.

Knackster, 2023-12-29:
I just subscribed to One Call for 3.0 so we shall see what happens in a few
hours.

Chris

henkdegroot, 2023-12-30:
> If you signed up for the 3.0 One Call, it should work with the 2.5 One Call that PiClock uses.

Just to confirm, I did sign up for the 3.0 One Call and it the repsonse I got was access denied. However, it seems that has been changed now. As when I try it today, it still works. So no update seems to be needed.

Perhaps some hint in the install documenation that this "error" can be ignored when you do not sign up for the One Call option, as it will then fallback to the free options for forecast and current. Being new to this, I did not realize the main difference is just the fact that One Call has all data in One Call, while forecast and current are two separate ones. I thought there would be more functionatlity that I missed.

Now hoping I get the Rainviewer data display correct for my area, seems to be a blindspot where I am located as I have never seen clouds in radar2 yet and in radar1 I do see cloulds but also not at my location. Anyway that is a different topic.

SerBrynden, 2023-12-30:
When you sign up with OpenWeather or make changes to a subscription plan, their emails mention how it can take a couple hours for the changes to take effect.

I've tweaked the log messages and changed them from an error to a warning.