n0bel/PiClock #265: Raspberry Pi OS with desktop Debian version: 12 (bookworm)

lstew452a, 2023-10-13:
I tried to install PiClock on the new Raspberry Pi OS with desktop Debian version: 12 (bookworm) and at step  install qt4 for python with command using superuser mode apt-get install python-qt4. I got an error Unable to locate package python-qt4. The OS is running Python 3.11.2 . Please advise.

SerBrynden, 2023-10-13:
Qt4 isn't available in Python3, and hasn't been available in Raspberry Pi OS for awhile. Try the instructions in this other branch: https://github.com/n0bel/PiClock/tree/Python3-SerBrynden or at this fork, which is the most up-to-date: https://github.com/SerBrynden/PiClock.
However, Bookworm is very new and we have yet to test PiClock with it.

SerBrynden, 2023-11-12:
@lstew452a This fork is up to date and works with Bookworm now: https://github.com/SerBrynden/PiClock

SerBrynden, 2023-11-15:
Recommend closing this issue.