n0bel/PiClock #267: SerBrynden Fork of PiClock

maserowik, 2023-11-12:
@SerBrynden 

      Not sure if this is the correct forum for th eFork of SerBrynden PiClock but I just intalled that version and  trying to use the switcher.sh command to switch different screen but i am getting a error

[2023-11-12 15:27:20.369123] Traceback (most recent call last):
[2023-11-12 15:27:20.399379] File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 16, in <module>
[2023-11-12 15:27:20.443457] import dateutil.parser
[2023-11-12 15:27:20.473403] ModuleNotFoundError: No module named 'dateutil'

this only happen I do the following ssh commands
./switcher.sh /home/pi/PiClock/Config-Noon

-or -
 
/home/pi/PiClock/switcher.sh Config-Noon

Thanks
Mike

SerBrynden, 2023-11-12:
Probably because the switcher.sh is not starting the virtual environment.  It is updated now.

maserowik, 2023-11-12:
So will a git pull work or do I need to reinstall?

Sent from the all new AOL app for iOS

SerBrynden, 2023-11-12:
Just `git pull`

maserowik, 2023-11-12:
I did a git pull and tried the switcher.sh again got same message in the log file.
[2023-11-12 18:28:03.154960] Traceback (most recent call last):[2023-11-12 18:28:03.175041] File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 16, in <module>[2023-11-12 18:28:03.205056] import dateutil.parser[2023-11-12 18:28:03.235686] ModuleNotFoundError: No module named 'dateutil'

SerBrynden, 2023-11-13:
You're missing the python-dateutil package.  Did you follow the instructions at this link, which have been updated for Bookworm? https://github.com/SerBrynden/PiClock/blob/master/Documentation/Install.md

maserowik, 2023-11-13:
yes that is the instuction I followed.  Is the python-dateutil part of what package?
***@***.***:/etc $ cat os-releasePRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"NAME="Raspbian GNU/Linux"VERSION_ID="11"VERSION="11 (bullseye)"VERSION_CODENAME=bullseyeID=raspbianID_LIKE=debianHOME_URL="http://www.raspbian.org/"SUPPORT_URL="http://www.raspbian.org/RaspbianForums"BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

SerBrynden, 2023-11-13:
It's in the requirements.txt

maserowik, 2023-11-13:
python3 -m pip install -r requirements.txt
rerunning this command.

maserowik, 2023-11-13:
Brendan,    I redid the installation and still facing the same issue as before

[2023-11-13 17:00:03.923178] Traceback (most recent call last):[2023-11-13 17:00:03.953013] File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 16, in <module>[2023-11-13 17:00:03.982998] import dateutil.parser[2023-11-13 17:00:04.023742] ModuleNotFoundError: No module named 'dateutil'

i am running on a pi zero w incase that matters.

SerBrynden, 2023-11-13:
Can you send me what you have installed using these commands?
```
python3 --version
cd "$HOME"/PiClock
source venv/bin/activate
pip list
```

maserowik, 2023-11-14:
@SerBrynden

***@***.***:~ $ python3 --versionPython ***@***.***:~ $ cd ***@***.***:~/PiClock $ source venv/bin/activate(venv) ***@***.***:~/PiClock $ pip listPackage           Version----------------- --------------arandr            0.1.10astroid           2.5.1asttokens         2.0.4automationhat     0.2.0beautifulsoup4    4.9.3blinker           1.4blinkt            0.1.2buttonshim        0.0.2Cap1xxx           0.1.3certifi           2020.6.20cffi              1.16.0chardet           4.0.0click             7.1.2colorama          0.4.4colorzero         1.1cryptography      3.3.2cupshelpers       1.0dbus-python       1.2.16distro            1.5.0dnspython         2.0.0docutils          0.16drumhat           0.1.0envirophat        1.0.0ExplorerHAT       0.4.2Flask             1.1.2fourletterphat    0.1.0gpg               1.14.0-unknowngpiozero          1.6.2h3                3.7.6html5lib          1.1idna              2.10isort             5.6.4itsdangerous      1.1.0jedi              0.18.0Jinja2            2.11.3lazy-object-proxy 0.0.0logilab-common    1.8.1lxml              4.6.3Markdown          3.3.4MarkupSafe        1.1.1mccabe            0.6.1microdotphat      0.2.1mote              0.0.4motephat          0.0.3mypy              0.812mypy-extensions   0.4.3numpy             1.19.5oauthlib          3.1.0pantilthat        0.0.7parso             0.8.1pexpect           4.8.0pgzero            1.2phatbeat          0.1.1pianohat          0.1.0picamera          1.13picamera2         0.3.12pidng             4.0.9piexif            1.1.3piglow            1.2.5pigpio            1.78Pillow            8.1.2pip               23.3.1pkg_resources     0.0.0psutil            5.8.0pycairo           1.16.2pycparser         2.21pycups            2.0.1pygame            1.9.6Pygments          2.7.1PyGObject         3.38.0pyinotify         0.9.6PyJWT             1.7.1pylint            2.7.2PyOpenGL          3.1.5pyOpenSSL         20.0.1PyQt5             5.15.2PyQt5-sip         12.8.1pyserial          3.5b0pysmbc            1.0.23python-apt        2.2.1python-dateutil   2.8.2python-metar      1.4.0python-prctl      1.7pytz              2023.3.post1PyYAML            5.3.1rainbowhat        0.1.0reportlab         3.5.59requests          2.25.1requests-oauthlib 1.0.0requests-toolbelt 0.9.1responses         0.12.1roman             2.0.0RPi.GPIO          0.7.0RTIMULib          7.2.1scrollphat        0.0.7scrollphathd      1.2.1Send2Trash        1.6.0b1sense-hat         2.6.0setuptools        68.2.2simplejpeg        1.6.4simplejson        3.17.2six               1.16.0skywriter         0.0.7sn3218            1.2.7soupsieve         2.2.1spidev            3.5ssh-import-id     5.10thonny            4.0.1timezonefinder    6.2.0toml              0.10.1touchphat         0.0.1twython           3.8.2typed-ast         1.4.2typing-extensions 3.7.4.3tzlocal           5.2unicornhathd      0.0.4urllib3           1.26.5v4l2-python3      0.3.2w1thermsensor     2.3.0webencodings      0.5.1Werkzeug          1.0.1wheel             0.34.2wrapt             1.12.1(venv) ***@***.***:~/PiClock $
if remote access is needed I can set that up.
Mike

maserowik, 2023-11-14:
[@SerBrynden]
(https://github.com/SerBrynden)

 python3 --version
Python 3.9.2
pi@pi-clock:~ $ cd "$HOME"/PiClock
pi@pi-clock:~/PiClock $ source venv/bin/activate
(venv) pi@pi-clock:~/PiClock $ pip list
Package           Version
----------------- --------------
arandr            0.1.10
astroid           2.5.1
asttokens         2.0.4
automationhat     0.2.0
beautifulsoup4    4.9.3
blinker           1.4
blinkt            0.1.2
buttonshim        0.0.2
Cap1xxx           0.1.3
certifi           2020.6.20
cffi              1.16.0
chardet           4.0.0
click             7.1.2
colorama          0.4.4
colorzero         1.1
cryptography      3.3.2
cupshelpers       1.0
dbus-python       1.2.16
distro            1.5.0
dnspython         2.0.0
docutils          0.16
drumhat           0.1.0
envirophat        1.0.0
ExplorerHAT       0.4.2
Flask             1.1.2
fourletterphat    0.1.0
gpg               1.14.0-unknown
gpiozero          1.6.2
h3                3.7.6
html5lib          1.1
idna              2.10
isort             5.6.4
itsdangerous      1.1.0
jedi              0.18.0
Jinja2            2.11.3
lazy-object-proxy 0.0.0
logilab-common    1.8.1
lxml              4.6.3
Markdown          3.3.4
MarkupSafe        1.1.1
mccabe            0.6.1
microdotphat      0.2.1
mote              0.0.4
motephat          0.0.3
mypy              0.812
mypy-extensions   0.4.3
numpy             1.19.5
oauthlib          3.1.0
pantilthat        0.0.7
parso             0.8.1
pexpect           4.8.0
pgzero            1.2
phatbeat          0.1.1
pianohat          0.1.0
picamera          1.13
picamera2         0.3.12
pidng             4.0.9
piexif            1.1.3
piglow            1.2.5
pigpio            1.78
Pillow            8.1.2
pip               23.3.1
pkg_resources     0.0.0
psutil            5.8.0
pycairo           1.16.2
pycparser         2.21
pycups            2.0.1
pygame            1.9.6
Pygments          2.7.1
PyGObject         3.38.0
pyinotify         0.9.6
PyJWT             1.7.1
pylint            2.7.2
PyOpenGL          3.1.5
pyOpenSSL         20.0.1
PyQt5             5.15.2
PyQt5-sip         12.8.1
pyserial          3.5b0
pysmbc            1.0.23
python-apt        2.2.1
python-dateutil   2.8.2
python-metar      1.4.0
python-prctl      1.7
pytz              2023.3.post1
PyYAML            5.3.1
rainbowhat        0.1.0
reportlab         3.5.59
requests          2.25.1
requests-oauthlib 1.0.0
requests-toolbelt 0.9.1
responses         0.12.1
roman             2.0.0
RPi.GPIO          0.7.0
RTIMULib          7.2.1
scrollphat        0.0.7
scrollphathd      1.2.1
Send2Trash        1.6.0b1
sense-hat         2.6.0
setuptools        68.2.2
simplejpeg        1.6.4
simplejson        3.17.2
six               1.16.0
skywriter         0.0.7
sn3218            1.2.7
soupsieve         2.2.1
spidev            3.5
ssh-import-id     5.10
thonny            4.0.1
timezonefinder    6.2.0
toml              0.10.1
touchphat         0.0.1
twython           3.8.2
typed-ast         1.4.2
typing-extensions 3.7.4.3
tzlocal           5.2
unicornhathd      0.0.4
urllib3           1.26.5
v4l2-python3      0.3.2
w1thermsensor     2.3.0
webencodings      0.5.1
Werkzeug          1.0.1
wheel             0.34.2
wrapt             1.12.1
(venv) pi@pi-clock:~/PiClock $ pi@pi-clock:

SerBrynden, 2023-11-14:
Looks like the packages are fine.  Run your switcher.sh in your terminal again and show me the messages from both the terminal and the log file.

maserowik, 2023-11-14:
Last login: Tue Nov 14 06:25:47 2023 from 192.168.55.157
pi@pi-clock:~ $  sh /home/pi/PiClock/switcher.sh Config
Activating virtual environment...
/home/pi/PiClock/switcher.sh: 6: source: not found
Rotating log files...
Starting PiClock... logging to Clock/PyQtPiClock.1.log

@SerBrynden 

[2023-11-14 06:31:07.232970] Traceback (most recent call last):
[2023-11-14 06:31:07.261023] File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 16, in <module>
[2023-11-14 06:31:07.293083] import dateutil.parser
[2023-11-14 06:31:07.322995] ModuleNotFoundError: No module named 'dateutil'

SerBrynden, 2023-11-14:
```
pi@pi-clock:~ $ sh /home/pi/PiClock/switcher.sh Config
Activating virtual environment...
/home/pi/PiClock/switcher.sh: 6: source: not found
```
There is the first error: `/home/pi/PiClock/switcher.sh: 6: source: not found` 
The `sh` command is not working with the virtual environment.  
Use `bash /home/pi/PiClock/switcher.sh Config` instead.

maserowik, 2023-11-15:
> ```
> pi@pi-clock:~ $ sh /home/pi/PiClock/switcher.sh Config
> Activating virtual environment...
> /home/pi/PiClock/switcher.sh: 6: source: not found
> ```
>
> There is the first error: [...]

so far this appears to be working.  I will close this tomorrow.

Thanks
Mike

maserowik, 2023-11-15:
@SerBrynden 
Thank you for your assistance.  I have everything backup and working!!!!

closed 2023-11-15 by maserowik