n0bel/PiClock #268: Error starting Pi-Clock

maserowik, 2023-11-21:
@SerBrynden 

Your fork has been working correctly until today.
I have my clock restart with different screen through out the day
today when the clock started I see the clock load but then the clock stops
looking at the log file I see the following

[2023-11-21 07:13:35.245346-05:00] File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 1284, in wxfinished_metar
[2023-11-21 07:13:35.359718-05:00] wd = f.wind_dir.compass()
[2023-11-21 07:13:35.463263-05:00] AttributeError: 'NoneType' object has no attribute 'compass'

any ideas?

SerBrynden, 2023-11-21:
The wind direction is missing from the METAR weather conditions.  Was there any error in getting the METAR data?  Look for "wxmetar:" in the log.

maserowik, 2023-11-21:
2023-11-21 07:13:33.045764-05:00] radar3... get radar tiles for time 1700550600 (2023-11-21 02:10:00-05:00)
[2023-11-21 07:13:33.206202-05:00] radar3 tile0 https://tilecache.rainviewer.com/v2/radar/1700550600//256/7/33/46/6/1_1.png
[2023-11-21 07:13:33.585351-05:00] radar2... get radar tiles for time 1700550600 (2023-11-21 02:10:00-05:00)
[2023-11-21 07:13:33.764502-05:00] radar2 tile0 https://tilecache.rainviewer.com/v2/radar/1700550600//256/11/567/772/6/1_1.png
[2023-11-21 07:13:34.976032-05:00] wxmetar: KAFJ 211156Z AUTO VRB04KT 10SM OVC080 06/01 A3012 RMK AO2 RAE33 SLP206 P0000 60001 70001 T00560011 10061 20050 55009
[2023-11-21 07:13:35.143423-05:00] Traceback (most recent call last):
[2023-11-21 07:13:35.245346-05:00] File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 1284, in wxfinished_metar
[2023-11-21 07:13:35.359718-05:00] wd = f.wind_dir.compass()
[2023-11-21 07:13:35.463263-05:00] AttributeError: 'NoneType' object has no attribute 'compass'

SerBrynden, 2023-11-21:
Since the wind is variable at 4 knots, as given by the code "VRB04KT", there is no specific direction the wind is coming from.  This causes `None` to be returned by the `f.wind_dir.compass()` line.  I've updated PiClock to ignore the wind direction if it is `None`.  Just do a `git pull` to update.
I guess variable winds are rare in Metar reports, so this error hasn't been caught before.  Or, not many people are using Metar because it only updates every hour.

maserowik, 2023-11-21:
ok so i am NOT a github person. What does the following mean?  and how do i solve it?
  
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Total 4 (delta 3), reused 4 (delta 3), pack-reused 0
Unpacking objects: 100% (4/4), 1.06 KiB | 18.00 KiB/s, done.
From https://github.com/SerBrynden/PiClock
   e500856..3485510  master     -> origin/master
Updating e500856..3485510
error: Your local changes to the following files would be overwritten by merge:
        Clock/PyQtPiClock.py
Please commit your changes or stash them before you merge.
Aborting

SerBrynden, 2023-11-21:
You may have made changes to `PyQtPiClock.py`.  Try this to discard any local changes and force the update:
```
git reset --hard
git pull
```

maserowik, 2023-11-21:
Correct I made all kinds of changes to that file to customize to my liking.  is there a way I can keep my changes and apply you fix?

SerBrynden, 2023-11-21:
You can see the updates that were made here: https://github.com/SerBrynden/PiClock/commit/34855100cf8127793baae84a8f2d4ccebf46e0f4

maserowik, 2023-11-21:
Thank you.  All is back up and working.  Closing this issue.

closed 2023-11-21 by maserowik