n0bel/PiClock #269: More Metar issues.

maserowik, 2023-11-21:
@SerBrynden 
I spoke to soon  there is now another issue wiht the METAR Parsing.
I have remove the metar info for now.
 

[2023-11-21 14:03:31.565007] getting METAR current conditions: https://tgftp.nws.noaa.gov/data/observations/metar/stations/KAFJ.TXT
[2023-11-21 14:03:31.628588] getting OpenWeather One Call: https://api.openweathermap.org/data/2.5/onecall?appid=88f580498dfd90f214bea557f04a7b3a&lat=40.3044&lon=-80.0717&units=imperial&lang=en&r=0.4420004234164133
[2023-11-21 14:03:31.926712] wxmetar: KAFJ 211856Z AUTO VRB05KT 10SM RA OVC011 07/07 A2994 RMK AO2 CIG 007V015 SLP143 P0003 T00720067
[2023-11-21 14:03:32.025810] Traceback (most recent call last):
[2023-11-21 14:03:32.065692] File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 1291, in wxfinished_metar
[2023-11-21 14:03:32.106015] wd = f.wind_dir.compass()
[2023-11-21 14:03:32.145214] AttributeError: 'NoneType' object has no attribute 'compass'

SerBrynden, 2023-11-21:
Make sure the if statements from the recent update are correctly indented.

maserowik, 2023-11-21:
@SerBrynden 

I was unable to get it to work with modifying the file by itself. 
I did a 

git reset --hard
git pull

I just redid the changes to the PyQtPiClock.py for the settingsI needed.

Closing this.

closed 2023-11-21 by maserowik