n0bel/PiClock #272: Multiple Errors with PiClock and PiClock3

nsnriv10022, 2023-12-14:
Installed stretch as directed and then followed directions for installing PiClock. Stretch is no longer supported by raspberry pi so the updates and install files needed are not retrievable. As well as the needed python qt4 files.
I have tried using stretch and a current version of Pi-os and can't get it to fly. I have also tried the same thing with PiClock3 and this gets closer but comes up with a time zone error message and won't finishing booting. I know these projects haven't been visited in some time but hoping there is still enough interest in getting this going again. Ill help in any way I can.

SerBrynden, 2023-12-14:
This fork is the most up-to-date: https://github.com/SerBrynden/PiClock
I works with the latest Raspberry Pi OS, Python3, and qt5.  There instructions there are updated as well.

nsnriv10022, 2023-12-14:
Thank you so much I'll do down that rabbit hole. Gonna have to format this SD card and start over.

nsnriv10022, 2023-12-14:
@SerBrynden followed your instructions and come up with error of line 16 in module 
no module named dateutil.
anyway you could email me directly? that would be much appreciated!
nsnriv10022@yahoo.com

SerBrynden, 2023-12-14:
We can correspond on here.
I assume you get that error after going through the install steps and running PiClock.  
Were there any error messages when installing the python packages using `python3 -m pip install -r requirements.txt` while in the virtual environment?
How are you running PiClock?

nsnriv10022, 2023-12-14:
I was getting an error message with the requirements.txt command during the install on the last step.
Tried it twice, then dumped the sd card and started over and it went the third time on a fresh install.
It wants to boot, now just getting error messages due to no api keys, which obviously I know why because I didn't enter any yet, to make sure the other errors were gone.

I had questions with regard to your other projects also but nothing to do with the clock. Don't want to clog this post up with unrelated discussion.

Thank you for your help on the clock

SerBrynden, 2023-12-14:
Don't worry about clogging up this post, but you can always open new discussions or issues on the other projects.
When it comes to installing the requirements.txt, make sure you're following all the instructions in detail at https://github.com/SerBrynden/PiClock/blob/master/Documentation/Install.md
You should have a virtual environment setup in the PiClock folder, and to install the python packages from requirements.txt as well as running PiClock, you need to be in the virtual environment by using the activate command.  All of which is in the instructions.
You can paste any error messages or log output in this post, just remember to strip out any API keys or other personal information.

nsnriv10022, 2023-12-14:
Do I have to have it pull from either google or mapbox? Can I have it pull a map from another website?
I am in a 2 way radio club, I would like the clock to display our linked repeater map instead of one of the other maps.
Is this possible without a major overhaul of the program?

SerBrynden, 2023-12-14:
Just Google or Mapbox for now. Does your other map provider have an API? I could look into it. But yes, the PiClock code would have to be changed.

nsnriv10022, 2023-12-14:
http://stats.nggmrs.net:8000/

SerBrynden, 2023-12-14:
That webpage won't work.  It's not a map provider.  It's just overlaying repeater locations onto Google Maps, with no API or information about how other users could do the same in another application.

nsnriv10022, 2023-12-14:
Alright sir that's what I was afraid you would say. I don't feel comfortable putting a credit card number in for an api key that I dont even intend to use. I will play with it some more and see if I can get that linked repeater map to work. Really I'm looking for the clock, weather, and linked repeater map functions. Not so concerned with radar, and other maps.

SerBrynden, 2023-12-14:
No problem. PiClock wasn't created to be customizable in that way, not without major overhaul. PiClock was designed to show the weather radar over regular maps. If you ARE ever interested in

nsnriv10022, 2023-12-15:
I got the clock working as designed, for future reference you do need a CC for Mapbox now. They wont even let you create an account without a CC. Got the image working with maps as you wrote the code thank you for your help. I will copy the IMG to my desk top and save that. Then I'll try to change the lower left radar images to instead the repeater map. Thank you again for all your help sir!