n0bel/PiClock #274: Pi Clock Weather Web Display

maserowik, 2024-01-04:
@SerBrynden 

[image]

How did you get the full USA to display on this?

Mike

SerBrynden, 2024-01-04:
Adjust the zoom level and coordinates for that particular radar frame in the config file.

maserowik, 2024-01-04:
do you happend to know what section on the in the code to change.  There a quite a few places in there.

SerBrynden, 2024-01-04:
In your config file: 
```python
radar4 = {
    'center': LatLng(38, -96),
    'zoom': 4,
    'basemap': map_base,
    'overlay': map_overlay,
    'color': 6,
    'smooth': 1,
    'snow': 1,
    'markers': (
        {
            'visible': 0,
            'location': radar_location,
            'color': 'red',
            'size': 'small',
            'image': 'teardrop-dot',
        },
    )
}
```

maserowik, 2024-01-05:
thanks closing this.

closed 2024-01-05 by maserowik