n0bel/PiClock #279: No weather display or data

godofbooks, 2024-11-27:
PiClock version: https://github.com/SerBrynden/PiClock/tree/master

Hardware: Raspberry Pi 0W, fresh install of 32 bit Raspberry Pi OS

Issue: Weather data not displaying at all (see attached image)
[image]

I followed the install instructions as written. API keys (OpenWeatherMaps & Mapbox) are working. Log file says:
[2024-11-26 21:47:57.628969] ERROR: Config file not found -n.py

I'm not sure what config file this is referring to or how to fix it. Clock will load, but no weather data is displayed.

vanhalen26505, 2024-11-27:
Mine started looking like this recently as well. It had been running for
many months. Interested to hear if there is a solution or update of some
kind.

Thanks,
Lee

SerBrynden, 2024-11-27:
I'm unable to recreate the problem.  When running PiClock manually from the terminal, be sure you are using `bash startup.sh -n -s` and not `bash switcher.sh -n -s` because the latter will not work with the `-n -s` arguments.

All I can suggest is to start over from scratch and follow the instructions again at https://github.com/SerBrynden/PiClock/blob/master/Documentation/Install.md and report any more problems after that.

godofbooks, 2024-11-27:
Will do. I'm at work at the moment, but I'll reinstall when I get home & update any issues on the correct fork. Thanks for the input!

closed 2024-11-27 by godofbooks