n0bel/PiClock #280: Radar stopped displaying - OpenWeather

Barry-IA, 2025-09-19:
Hi!
As of about two days ago (so around September 16, 2025) my radar stopped displaying: no time, no coloured blobs, just the map that is supposed to be the underlay.  I am getting updated forecasts.  I didn't change any configurations, thought of course the 'behind the scenes' ones  My Raspberry Pi does say it's up to date.

OpenWeatherMap says my API is active -- which of course it is because I get the forecasts.

I have restarted PiClock.
I didn't see anything in /var/logs, but I'm also not finding a current log for PiClock -- I need t find how to re-activate that critter.
Thanks for any hints!
Barry

SerBrynden, 2025-09-19:
There is a problem in PiClock that has existed for years, but is only now causing problems.  You need to remove the extraneous slash in the RainViewer API call URL.  This used to be ignored by RainViewer but now causes an invalid response.

In `PyQtPiClock.py`, search for the two lines that begin with `tail = "/256/%d/%d/%d` and remove the slash before `256`.

[image]

SerBrynden, 2025-09-19:
And the PiClock logs are in the `PiClock/Clock/` folder -> https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#logs

Knackster, 2025-10-23:
Just wanted to comment that this fix worked for me as well. I've been in New Mexico for 3 month and when I got back a few weeks ago I didn't realize it wasn't working until a storm came through last weekend . Thanks team.

ScottChapman, 2025-11-23:
> I'm using OWM so it doesnt affect me. Its working just fine.

Are you using OWM for radar images? How?